mathisschmieder/HL2-MRF101 #2: K12 logic incorrect

m5evt, 2020-11-21:
Rx signals are only present by removing FB3. I can hear RX signals through RF3. In changing to the 12 V AXICOM relays, I believe the logic of the main TX/RX relay K12 has been incorrectly assigned (by me!). 

The hardware modification of cutting tracks and swapping the relay pin sets is not too appealing. I am proposing that a logic inverter (sn74ahc1gu04) and a NUD3124 are connected between the pads of FB3. Does anyone have any better solutions for already purchased PCBs? 

I also notice that without this mod, Q2 gate will be pulled up to 12 V. In my PR I set the option for EXTTR or INTTR to be used for PTT. INTTR is pulled up to 3V3 through HL2 relay K2.  EXTTR has no pull up. A pull up resistor footprint should probably be placed on this board in a future revision for when EXTTR is used?

m5evt, 2020-11-22:
Fresh pair of eyes after a good night of sleep :) Summarise the issue to; DNF D3 if using EXTTR.

As a side note, PTT out (J5) should maybe have an open drain output on a future revision?

m1geo, 2021-02-01:
Agree with @m5evt. Switched to use EXTTR.