mathisschmieder/HL2-MRF101 #3: FWD and REF power

m5evt, 2020-12-06:
R4 and R11 at 56k clips the op amp input at just over 100 W (provided my external power meter is accurate!). 47k for both of these gives a little bit of room to before reaching the 3v3 limit.

m5evt, 2021-03-07:
I have changed to 20 turns on an FT50A-37. FT50A-61 with 10 turns does not present a good impedance on 160m. See [here](https://groups.google.com/g/hermes-lite/c/baTuOzigmsw/m/xDoaNty4CwAJ) for further discussion.

m1geo, 2023-08-21:
Sorry to wake a closed ticket, but I cannot find any reference of FT50A-37. Could you please point me to a supplier. Not even eBay has them, which is what the BoM V0.4 lists as the source.

I'm currently using 10t:1t on FT50A-61 mix and one of the ferrites gets quite hot.

m5evt, 2023-09-17:
Try searching FT-50A-37 (or similar variants with the dash in different places). I think I got mine from Spectrum Communications on ebay.