mathisschmieder/HL2-MRF101 #5: Input power attenuator issues

m1geo, 2021-02-01:
The input power attenuator gets hot, and thus effects the thermal tracking of Q1.

The input power attenuator could easily be moved away from the bias tracking.

m5evt, 2021-02-01:
@m1geo I am currently looking at a more permanent housing for my HL2/PA combo. Thermal issues are the current headache. I see a higher ambient temperature on the HL2 reported temperature due to reduced airflow over the top (as the PA PCB is blocking it. The attenuator also heats up this gap between the HL2 and the PA. Granted a fan could reduce the problem, but it doesn't seem to be fixing it. I would like to get rid of the attenuator, but lack a spectrum analyser to confirm how much worse the IMD of the TX signal is at 1-2 W compared with 5 W. Do you have a SA to help characterise this and allow us to hopefully get rid of the attenuator?

mathisschmieder, 2021-02-11:
I do have access to spectrum analyzers at work (although I'm not often there at the moment...) and could do the measurements.

Some attenuation between the HL2 and the PA should be kept, at least IMHO, just as a safety net for when you accidentally transmit with 5W. It will happen, eventually ;)

Maybe we should think about re-designing and moving the attenuator pad and possibly also adding a small heat sink to it?

m5evt, 2021-03-07:
I have changed to a 3 dB attenuator. Heat is a lot less and monitoring the transmitted signal in SDR software with a 2 tone shows the IMD is acceptable at lower power from the HL2.