mathisschmieder/HL2-MRF101 #6: High noise on the ADC signals

m5evt, 2021-02-01:
I have modified my [linHPSDR fork](https://github.com/m5evt/linhpsdr) to read current and temperature from the ADC via the i2c read/write functions in the HL2 extended protocol. However, the noise on the ADC traces is very high. With heavy filtering in software the temperature can be read, but the current measurement have too much noise and does not allow the 100 mA bias current to be set accurately.

m1geo, 2023-08-16:
I'm building a derivative of this board and can confirm that the ADC sees high noise when there is RF on the board. The ADC can set the 100mA static current (as there's no RF) but as soon as RF at any frequency is applied, then there is considerable ADC noise. If I fix this, I will report back.

m1geo, 2023-08-20:
I found that decreasing the I2C resistors a bit (to around 700 Ohms total) helped quite a bit. However, you cannot go too low as the ADC cannot sink enough current. I'm using an onboard STM32 which is happy at 1.7 MHz I2C SCL in such configuration.

m5evt, 2023-09-17:
You have probably already seen this referenced in another closed ticket in this repo https://github.com/m5evt/hl2_extpamon. I removed the ADC and use this board instead now. Zero issues. I think keeping this as a separate board away from the 100 W RF is sensible. I have blown up two MRF101 FETs with thermal issues (before making this board). I now crudely cut the PA 48 V if I > 3A. A better control system for someone up for the challenge should adjust the pa bias based on the measured temperature.