mathisschmieder/HL2-MRF101 #7: Add small microcontroller on the PCB

m5evt, 2021-02-01:
The HL2 does not have any gateware code to talk to the ADC (to report PA current and PA temperature on the PC), this must be done via the i2c read/write functions in the [protocol](https://github.com/softerhardware/Hermes-Lite2/wiki/Protocol). These read/writes starve the normal C&C responses from the HL2. A small micro controller on the PCB could handle the ADC measurements and reduce "starving" the HL2 protocol 1 C&C responses. 

A small micro controller could sit on the HL2 i2c2 bus and talk to the ADC on a separate i2c bus, within the micro filtering, power calibration tables etc. could be held. Or remove the ADC and use a micro with built in ADC?

m5evt, 2021-02-01:
This could also control fans specifically for the PCB. Note, at the moment there is no way to control the HL2 fan output via the protocol. The fan will only be turned on if the HL2 temperature sensor (next to the HL2 5 W PA) goes above a certain value. This micro controller would allow better control of the hl2-mrf101 thermal profile.

mathisschmieder, 2021-02-11:
Sounds like a very reasonable idea and we should have the space on the PCB. I'm a fan of STM32, but open to other MCUs. Do you have a preference?

m5evt, 2021-03-07:
I've always used the AVRs but I felt like learning something new so I recently ordered an STM32 dev board. I'm going to prototype the idea on a separate PCB then when something is working, it can maybe be transferred to the HL2-MRF101 PCB.