mathisschmieder/HL2-MRF101 #8: Move zener D4

m5evt, 2021-03-07:
GW2HFR pointed out the thermal tracking over compensates. The main problem is the location of the zener diode D4.

I moved the zener D4 to the the drain of Q2. There is a ground via perfectly located nearby to solder the other end to. This is the least heat affected part of this bit of the PCB. I then removed R29 and put D5 from the wiper of the digi pot to ground. Thermal testing shows this is much less aggressive in compensating.

This should be changed in a future revision of the board.

m1geo, 2023-04-19:
@m5evt is there anything special about the PDZ5.1BGWJ over any other 5.1V Zener diode? Can I substitute a through-hole part and thermal paste it (along with the 1N4148) to the MRF101? I'm using snippets of this design for another purpose, and wanted to learn from the bugs here! :)

m5evt, 2023-09-17:
Just check the temperature coefficient if substituting. However, this control system is primitive. See suggestion here re: adjusting bias based on measured temperature https://github.com/mathisschmieder/HL2-MRF101/issues/6